umati/Dashboard-OPCUA-Client #150: Unable to connect to Sample-Server of flatglass

karstenroethig, 2022-04-21:
I'm using the Sample-Server from here: https://github.com/umati/Glass-Plugfest/tree/main/SampleServer
It runs at localhost:48030 and I'm able to connect with UaExpert and Prosys UPC UA Browser.

My configuration file looks like this:

	{
	  "MachineCacheFile": "MachineCache_datahub.json",
	  "ObjectTypeNamespaces": [
		"http://opcfoundation.org/UA/",
		"http://opcfoundation.org/UA/IA/",
		"http://opcfoundation.org/UA/DI/",
		"http://opcfoundation.org/UA/Machinery/",
		"http://opcfoundation.org/UA/Machinery/Result/",
		"http://opcfoundation.org/UA/MachineTool/",
		"http://opcfoundation.org/GMS/",
		"http://opcfoundation.org/UA/Glass/Flat/"
	  ],
	  "NamespaceInformations": [
		{
		   "Namespace":"http://opcfoundation.org/UA/Glass/Flat/",
		   "Types":[
			 {
			   "Uri":"http://opcfoundation.org/UA/Glass/Flat/",
			   "Id":"i=1015",
			   "$comment":"GlassMachineType"
			 }
		   ],
		   "IdentificationType":{
			 "Uri":"http://opcfoundation.org/UA/Glass/Flat/",
			 "Id":"i=1020",
			 "$comment":"GlassMachineIdentificationType"
		   }
		}
	  ]
	  ,
	  "OpcUa": {
		"Endpoint": "opc.tcp://localhost:48030",
		"Username": "",
		"Password": "",
		"Security": 1
	  },
	  "Mqtt": {
		"Hostname": "localhost",
		"Port": 1883,
		"Username": "opcua_client",
		"Password": ""
	  }
	}

After starting the client, I receive the following output:

	2022-04-21 13:52:01,227 [DashboardOpcUaClient] INFO void Umati::Util::ConfigureLogger(const string&):44 Logger Configured: DashboardOpcUaClient
	2022-04-21 13:52:01,228 [DashboardOpcUaClient] INFO int main(int, char**):60 Start Dashboard OPC UA Client
	2022-04-21 13:52:01,246 [DashboardOpcUaClient] WARNING Umati::OpcUa::Subscription::Subscription(const std::map<std::__cxx11::basic_string<char>, short unsigned int>&, const std::map<short unsigned int, std::__cxx11::basic_string<char> >&):50 Created subscription 0x55def159d4e8
	[2022-04-21 13:52:01.300 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
	[2022-04-21 13:52:01.314 (UTC+0200)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
	[2022-04-21 13:52:01.357 (UTC+0200)] info/securitypolicy        The basic256 security policy with openssl is added.
	[2022-04-21 13:52:01.357 (UTC+0200)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
	[2022-04-21 13:52:01.358 (UTC+0200)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
	[2022-04-21 13:52:01.359 (UTC+0200)] error/userland     Error Empty Certificate
	[2022-04-21 13:52:01.359 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
	[2022-04-21 13:52:01.359 (UTC+0200)] error/userland     Empty comparing subjectURI and applicationURI
	[2022-04-21 13:52:01.359 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
	[2022-04-21 13:52:01.359 (UTC+0200)] error/userland     Empty comparing subjectURI and applicationURI
	[2022-04-21 13:52:01.359 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
	[2022-04-21 13:52:01.359 (UTC+0200)] error/userland     Empty comparing subjectURI and applicationURI
	[2022-04-21 13:52:01.359 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
	[2022-04-21 13:52:01.359 (UTC+0200)] error/userland     Empty comparing subjectURI and applicationURI
	[2022-04-21 13:52:01.359 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
	[2022-04-21 13:52:01.364 (UTC+0200)] info/eventloop     Starting the EventLoop
	[2022-04-21 13:52:01.365 (UTC+0200)] warn/network       Connection to opc.tcp://localhost:48030 failed with error: Connection refused
	[2022-04-21 13:52:01.365 (UTC+0200)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadDisconnect
	2022-04-21 13:52:01,365 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):40 The client is disconnected
	2022-04-21 13:52:01,366 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):60 Session disconnected
	2022-04-21 13:52:01,367 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):68 Bad Disconnect
	2022-04-21 13:52:01,367 [DashboardOpcUaClient] ERROR bool Umati::OpcUa::OpcUaClient::connect():127 Connecting failed in OPC UA Data Client: BadDisconnect
	2022-04-21 13:52:01,367 [DashboardOpcUaClient] ERROR void Umati::OpcUa::OpcUaClient::connectionStatusChanged(UA_Int32, UA_ServerState):464 Disconnected.
	2022-04-21 13:52:01,376 [DashboardOpcUaClient] INFO Umati::MqttPublisher_Paho::MqttPublisher_Paho::MqttPublisher_Paho(const string&, uint16_t, const string&, const string&):27 Connect to localhost
	2022-04-21 13:52:01,485 [DashboardOpcUaClient] INFO virtual void Umati::MqttPublisher_Paho::MqttPublisher_Paho::MqttCallbacks::connected(const string&):98 Mqtt Connected: connect onSuccess called
	2022-04-21 13:52:01,485 [DashboardOpcUaClient] INFO bool DashboardOpcUaClient::connect(std::atomic_bool&):40 Waiting for OPC UA connection.
	2022-04-21 13:52:02,486 [DashboardOpcUaClient] INFO bool DashboardOpcUaClient::connect(std::atomic_bool&):40 Waiting for OPC UA connection.
	2022-04-21 13:52:03,491 [DashboardOpcUaClient] INFO bool DashboardOpcUaClient::connect(std::atomic_bool&):40 Waiting for OPC UA connection.

What's the issue here?
Is there something, thats needs to be done in the Sample-Server implementation?

ccvca, 2022-04-21:
>[2022-04-21 13:52:01.365 (UTC+0200)] warn/network       Connection to opc.tcp://localhost:48030 failed with error: Connection refused

Liefert die Ausgabe vom Glass-Server irgendwelche Hinweise? "Connection Refused" klingt danach, dass es in Server ne Exception gegeben hat.

Kantiran91, 2022-04-25:
We need to build the sample server with open62541 warning loglevel

karstenroethig, 2022-05-06:
I figured it out. The problem was network access between the Windows host and WSL.
In case someone has similar problems, this helps: https://docs.microsoft.com/en-us/windows/wsl/networking

closed 2022-05-06 by karstenroethig