umati/Dashboard-OPCUA-Client #376: Connection through a TLS inspection fw

GoetzGoerisch, 2023-03-10:
Using the gateway in `wss` mode through a TLS inspection firewall fails with:

`Paho Exception:MQTT error [-3], Disconnected`

Adding the inspection fw cert to `certs` on Windows changes the output to

`Paho Exception:MQTT error [5]: CONNACK return code`

As reference using MQTT Explorer the user needed to add the root cert to the server certificate (CA) option and it worked:
[image]

wlkrm, 2023-03-14:
Have we any change to get a similar test setup?

wlkrm, 2023-03-14:
@mohitkumaragarwal, here is a test script, please evaluate if this works behind your Firwall:

```python
import time
import paho.mqtt.client as paho
import ssl

#define callbacks
def on_message(client, userdata, message):
  # print("log:  received message")
  pass

def on_log(client, userdata, level, buf):
  print("log: ",buf)

def on_connect(client, userdata, flags, rc):
  client.subscribe('umati/#')

client=paho.Client(transport="websockets") 
client.on_message=on_message
client.on_log=on_log
client.on_connect=on_connect
client.on_message=on_message
client.ws_set_options(path="/ws")
print("connecting to broker")
client.tls_set("/etc/ssl/certs/ca-certificates.crt")
client.connect("dev.umati.app", 443, 60)

##start loop to process received messages
client.loop_start()
#wait to allow publish and logging and exit
time.sleep(2)
```

If successful, the output should look sth. like that:

```
connecting to broker
log:  Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b''
log:  Received CONNACK (0, 0)
log:  Sending SUBSCRIBE (d0, m1) [(b'umati/#', 0)]
log:  Received SUBACK
log:  Received PUBLISH (d0, q0, r1, m0), 'umati/opcUaToMqttOnline', ...  (1 bytes)
```

mohitkumaragarwal, 2023-03-14:
@wlkrm 

I get the following output after setting the certificate path to my root certificate. Please let me know what is the next step? Do I need to download a later version of the Dashboard client to test the flow?

[image]

wlkrm, 2023-03-14:
Are you using the latest version of the client (development branch)?
Are you referencing the correct Path to the certs folder in the config?

wlkrm, 2023-03-14:
You can also try adding the explicit config to the configuration.json:

```json
{ 
   ...
   "Mqtt": {
       "CaCertPath":"<path>/<to>/certs",
       "CaTrustStorePath": "<path>/<to>/cacert.pem",

       ...
   }
   ...
}
```

mohitkumaragarwal, 2023-03-14:
@wlkrm 

- I am using the following version on Windows 11: UmatiDashboardOpcUaClient-Release-windows-2022-amd64.zip
- My root certificate is placed in /bin/certs/**ca_cert_zscaler.crt**
- The /bin/certs already has cacert.pem file. I have also tried to copy the content of our root certificate inside the cacert.pem file as suggested earlier by Götz 
- I am executing the DashboardClient.exe from inside the bin folder from an Adminstrator console.

mohitkumaragarwal, 2023-03-14:
> You can also try adding the explicit config to the configuration.json:
> 
> ```json
> { 
>    ...
>    "Mqtt": {
>        "CaCertPath":"<path>/<to>/certs",
>        "CaTrustStorePath": "<path>/<to>/cacert.pem",
> 
>        ...
>    }
>    ...
> }
> ```

Ok, I will try this. Regarding the Python Test Script output, does it mean it is working?

mohitkumaragarwal, 2023-03-14:
@wlkrm 
I tried the option to give the certificate path in the configuration file, but it looks it did not work. Sharing the copy of the required files in the following zip.
[bin.zip](https://github.com/umati/Dashboard-OPCUA-Client/files/10967585/bin.zip)

wlkrm, 2023-03-14:
I cannot identify any issue, might be a bug in the cpp bindings for paho-c

wlkrm, 2023-03-14:
It would be pretty helpful if I had access to a test setup. As I cannot reproduce nor debug the issue right now. Any ideas?

mohitkumaragarwal, 2023-03-14:
@wlkrm  The best I could do from my side is having a remote session with you where I have the setup on my system. Will that work for you?

wlkrm, 2023-03-15:
I think giving me remote access e.g. via Visual Studio Code Live Share could be a solution, as a Meeting would be very inefficient, I guess.

mohitkumaragarwal, 2023-03-15:
@wlkrm I can provide access to my machine around 30 minutes anytime today from 12:00 to 16:00. We can have a quick call where I can show the setup. Will that work?

wlkrm, 2023-03-15:
I think that time slot is too short. I am creating a minimal test app similar to the python script and reach out to you via e-mail.

mohitkumaragarwal, 2023-03-15:
@wlkrm I will wait for your test app and try that. Later, if needed, will 1 hour of time with my machine be good? As it is my primary office machine, we can start with 1 hour of troubleshooting and if needed, can take more time based on the analysis.

wlkrm, 2023-03-17:
Shure, maybe you could set up a working IDE with debugging enabled in advance?
The source code (`WssTroubleshooter.cpp`): 
https://github.com/umati/Dashboard-OPCUA-Client/tree/troubleshooter

Setup with vs code:
```
mkdir _install
mkdir _deps
mkdir _build
cd _deps
cmake ../.github/ -DCMAKE_INSTALL_PREFIX:PATH=<PATH/TO/>Dashboard-OPCUA-Client/_install/ -DCMAKE_BUILD_TYPE=Debug -DPAHO_WITH_SSL=1
cmake --build .
```

`settings.json` for vs code:
```json
{
    "cmake.generator": "Visual Studio 17 2022",
    "cmake.buildDirectory": "${workspaceFolder}/_build",
    "cmake.installPrefix": "${workspaceFolder}/_install",
    "cmake.sourceDirectory": "${workspaceFolder}",
    "cmake.debugConfig": {
        "externalConsole": true,
        "args": [
            "configuration.json"
        ],
        "cwd": "${workspaceFolder}/_install/bin",
        "environment": [
      
        ]
    }
}
```

`configuration.json` (inside ${workspaceFolder}/_install/bin):
```json
{
  "MachineCacheFile": "MachineCache_datahub.json",
  "MachinesFilter": [],
  "ObjectTypeNamespaces": [
    "http://opcfoundation.org/UA/",
    "http://opcfoundation.org/UA/IA/",
    "http://opcfoundation.org/UA/IA-Prototyping/",
    "http://opcfoundation.org/UA/DI/",
    "http://opcfoundation.org/UA/Machinery/",
    "http://opcfoundation.org/UA/Machinery/Result/",
    "http://opcfoundation.org/UA/MachineTool/",
    "http://opcfoundation.org/UA/MachineTool-Prototyping/",
    "http://opcfoundation.org/UA/PlasticsRubber/GeneralTypes/",
    "http://opcfoundation.org/UA/PlasticsRubber/IMM2MES/",
    "http://opcfoundation.org/UA/PlasticsRubber/TCD/",
    "http://opcfoundation.org/UA/PlasticsRubber/HotRunner/",
    "http://opcfoundation.org/UA/PlasticsRubber/LDS/",
    "http://opcfoundation.org/UA/PlasticsRubber/Extrusion_v2/GeneralTypes/",
    "http://opcfoundation.org/UA/PlasticsRubber/Extrusion_v2/HaulOff/",
    "http://opcfoundation.org/UA/PlasticsRubber/Extrusion_v2/Corrugator/",
    "http://opcfoundation.org/UA/PlasticsRubber/Extrusion_v2/Extruder/",
    "http://opcfoundation.org/UA/PlasticsRubber/umati/generic/",
    "http://opcfoundation.org/UA/PlasticsRubber/umati/OPC40079ForUmati/",
    "http://opcfoundation.org/UA/PlasticsRubber/ImmToRobot/"
  ],
  "NamespaceInformations": [
    {
      "Namespace": "http://opcfoundation.org/UA/Woodworking/",
      "Types": [
        {
          "Uri": "http://opcfoundation.org/UA/Woodworking/",
          "Id": "i=2",
          "$comment": "WwMachineType"
        }
[... 135 lines cut ...]
       ],
       "IdentificationType":{
         "Uri":"http://opcfoundation.org/UA/Glass/Flat/",
         "Id":"i=1020",
         "$comment":"GlassMachineIdentificationType"
       }
    }
  ],
  "OpcUa": {
    "Endpoint": "opc.tcp://localhost:4840",
    "Username": "",
    "Password": "",
    "Security": 0
  },
  "Mqtt": {
    "Hostname": "dev.umati.app",
    "Port": 443,
    "Username": "opcua_client",
    "Password": "",
    "Prefix": "test",
    "clientId": "umati",
    "Protocol": "wss"
  }
}
```

mohitkumaragarwal, 2023-03-17:
@wlkrm Shall we try this now on my computer? If you are available, please let me know, I can give my computer to you for 1 hour now?

mohitkumaragarwal, 2023-03-17:
@wlkrm Please let me know if you are available now, we can have a short call for 10 mins and view the setup and you can use the setup to troubleshoot?

wlkrm, 2023-03-17:
Sadly no. Is sent you an Email with an alternative meeting

wlkrm, 2023-03-21:
I just found the solution to your problem. You are using the wrong credentials (the ones you provide in the configuration.json you sent me are for dev.umati.app and not the for umati.app). If you replace "Hostname": "umati.app" with "Hostname": "dev.umati.app" it should work fine. You have to specify the right username and password in the configuration. The python-script is working, because it uses an anonymous connection, which of course succeeds without publishing rights.

wlkrm, 2023-03-21:
@mohitkumaragarwal, if found your issue. Your credentials are simply wrong. I cannot connect to dev.umati.app with your credentials using the mqtt explorer.

mohitkumaragarwal, 2023-03-21:
@wlkrm @GoetzGoerisch 
Great, in that case, @GoetzGoerisch can help me with some valid credentials which I can try? 
or @wlkrm you have any test credentials for me to try right now?

mohitkumaragarwal, 2023-03-21:
@wlkrm @GoetzGoerisch 
After getting the updated credentials, the MQTT Explorer Connection Works... **(The Log-in Part)**
What is the next step? From my side, I have done the following:

Now, after running the Dashboard Client with the same configuration and as of now, it **DID NOT throw the previous errors.** 
[DashboardOpcUaClient-20230321.log](https://github.com/umati/Dashboard-OPCUA-Client/files/11026536/DashboardOpcUaClient-20230321.log)

After running the Dashboard Client, I have run the MQTT Explorer (with empty credentials), **I DO NOT see TighteningSystemType listed yet. Do I need to do something for it?**

GoetzGoerisch, 2023-03-21:
@mohitkumaragarwal please try again. Found an issue with the broker accesscontrollist on our side. Fixed now.

mohitkumaragarwal, 2023-03-21:
@GoetzGoerisch @wlkrm 
Thanks a lot for your support. One local OPC UA Server is working now. I will try 3 instances of the DashboardClient to connect to 3 different OPC UA Servers and Update the status...

**I guess, the final steps are:**
Add the respective root certificate from the company in the certs.pem file
Add the valid credentials in the configuration.json 
Connection should work...

mohitkumaragarwal, 2023-03-21:
@GoetzGoerisch I guess, to connect multiple OPC UA Servers, I need different ClientIds for atlas?
Two instances did not work... I am checking the details

wlkrm, 2023-03-22:
closed as resolved

closed 2023-03-22 by wlkrm