umati/Dashboard-OPCUA-Client #380: Client does not build type map, if connection instability during initial start

GoetzGoerisch, 2023-03-15:
Using server implementation:
https://gitlab.com/umati/infrastructure/TruMark

Log excerpt:
```
2023-03-15 10:23:56,582 [DashboardOpcUaClient] INFO void __cdecl Umati::Util::ConfigureLogger(const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &):44 Logger Configured: DashboardOpcUaClient
2023-03-15 10:23:56,583 [DashboardOpcUaClient] INFO int __cdecl main(int,char *[]):61 Start Dashboard OPC UA Client
2023-03-15 10:23:56,583 [DashboardOpcUaClient] WARNING __cdecl Umati::OpcUa::Subscription::Subscription(const class std::map<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,unsigned short,struct std::less<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > >,class std::allocator<struct std::pair<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const ,unsigned short> > > &,const class std::map<unsigned short,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct std::less<unsigned short>,class std::allocator<struct std::pair<unsigned short const ,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > &):50 Created subscription 0000000000365668
2023-03-15 10:23:56,602 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):42 Client is connecting
2023-03-15 10:23:56,604 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):75 Session disconnected
2023-03-15 10:23:56,606 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):48 Client hello sent
2023-03-15 10:23:56,608 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):75 Session disconnected
2023-03-15 10:23:56,611 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):60 Client open sent
2023-03-15 10:23:56,613 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):75 Session disconnected
2023-03-15 10:23:56,620 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):39 A SecureChannel to the server is open
2023-03-15 10:23:56,621 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):75 Session disconnected
2023-03-15 10:23:56,634 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):39 A SecureChannel to the server is open
2023-03-15 10:23:56,636 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):78 Session create requested
2023-03-15 10:23:56,647 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):39 A SecureChannel to the server is open
2023-03-15 10:23:56,649 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):81 Session created
2023-03-15 10:23:56,650 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):39 A SecureChannel to the server is open
2023-03-15 10:23:56,653 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):84 Session activate requested
2023-03-15 10:23:56,656 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):39 A SecureChannel to the server is open
2023-03-15 10:23:56,658 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::stateCallback(struct UA_Client *,UA_SecureChannelState,UA_SessionState,unsigned int):72 A session with the server is activated
2023-03-15 10:23:56,660 [DashboardOpcUaClient] ERROR void __cdecl Umati::OpcUa::OpcUaClient::connectionStatusChanged(int,UA_ServerState):449 Connected.

2023-03-15 10:23:56,663 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 0, namespaceURI: http://opcfoundation.org/UA/
2023-03-15 10:23:56,664 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 1, namespaceURI: urn:stg12_pc2_w7-64:Trumpf:UmatiUAServer
2023-03-15 10:23:56,667 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 2, namespaceURI: http://opcfoundation.org/UA/DI/
2023-03-15 10:23:56,670 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 3, namespaceURI: http://opcfoundation.org/UA/IA/
2023-03-15 10:23:56,672 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 4, namespaceURI: http://opcfoundation.org/UA/Machinery/
2023-03-15 10:23:56,675 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 5, namespaceURI: http://www.trumpf.com/UA/umati/
2023-03-15 10:23:56,677 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 6, namespaceURI: http://opcfoundation.org/UA/MachineTool/
2023-03-15 10:23:56,680 [DashboardOpcUaClient] INFO void __cdecl Umati::OpcUa::OpcUaClient::fillNamespaceCache(const class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > &):369 index: 7, namespaceURI: http://opcfoundation.org/UA/Diagnostics
2023-03-15 10:23:56,685 [DashboardOpcUaClient] ERROR void __cdecl Umati::OpcUa::Subscription::createSubscription(struct UA_Client *):98 Create subscription succeeded, id 2
2023-03-15 10:23:56,689 [DashboardOpcUaClient] INFO __cdecl Umati::MqttPublisher_Paho::MqttPublisher_Paho::MqttPublisher_Paho(const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,unsigned short,const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,const class std::basic_string [... cut ...]
2023-03-15 10:23:59,097 [DashboardOpcUaClient] ERROR class std::shared_ptr<class ModelOpcUa::StructureNode> __cdecl Umati::Dashboard::OpcUaTypeReader::typeDefinitionToStructureNode(const struct ModelOpcUa::NodeId_t &) const:357 Unable to find nsu=http://opcfoundation.org/UA/MachineTool/;i=13 in typeMap
2023-03-15 10:23:59,099 [DashboardOpcUaClient] WARNING void __cdecl Umati::Dashboard::OpcUaTypeReader::updateObjectTypeNames(void):98 Type not found. Probably because the server does not have the NodeId, specified in the config (nsu=http://opcfoundation.org/UA/MachineTool/;i=13).
```

Kantiran91, 2023-05-12:
@GoetzGoerisch Is this really a bug? how should we fix this?

wlkrm, 2025-02-11:
@GoetzGoerisch can this really be fixed?

GoetzGoerisch, 2025-02-12:
Don't know, as I have no understanding who is generating the problem here. Looks like that open62541 and .NET implemented servers have an issue.

wlkrm, 2025-02-12:
Can you please try with revert/431 container?

GoetzGoerisch, 2025-02-13:
Yes, will do, but only next week.
@ainglessi

GoetzGoerisch, 2025-02-18:
Still see this startup instablitiy but nevertheless for the time being I consider this as not fixable without deep debugging in both server and client.
Which is not worthwhile IMHO.

closed 2025-02-18 by GoetzGoerisch