umati/Dashboard-OPCUA-Client #410: Issue with installer on windows

altenstefanobranca, 2023-03-31:
I have discovered an error in the installation of the opcua-client. It appears to have been caused by a system configuration that is not compatible. I have attempted to install all four versions of the Windows installer for [Release 1.3.2](https://github.com/umati/Dashboard-OPCUA-Client/releases/tag/v1.3.2) in an effort to resolve the issue.
[image]
[image]

GoetzGoerisch, 2023-04-04:
@altenstefanobranca did you put a `configuration.json` into the `bin` folder, next to the `.exe`
Can you try run it from the command line please?
Please also try the latest 2.0.0-rc2 release.

@marcfir  are we missing any requirements to run on Windows 10 Enterprise 2016?

GoetzGoerisch, 2023-04-27:
@altenstefanobranca can you please try release 2.0.0-rc3

GoetzGoerisch, 2025-03-21:
Probably resolved? Please have a look at umatiGateway  @altenstefanobranca! Please create a new issue there if you still have an issue.