umati/Dashboard-OPCUA-Client #429: Issue with 32 bit Dashboard-OPCUA-Client

xenonforlife, 2023-04-14:
KUKA wants to connect their machine (implemented OPC UA for Robotics + OPC UA for Machinery) to the umati infrastructure. They tried with both the 32 bit binaries. They have added the fitting configuration.json within the bin folder and are getting the following error  (attached images) 
[image]
[image]

Here are the two Windows versions they have tried with:
1. Test-System: WES7V4.2.0 (32 Bit)
2. Windows on the Controller : Windows10IoTV5.1.0 (32 Bit)

BFrKUKA, 2023-04-17:
Test of both Win Clients (see screenshots) has been done at: Test-System with WES7V4.2.0 (32 Bit)

If it works on a Controller with Windows10IoTV5.1.0 (32 Bit) was not tested by my side -> but KUKA systems on Automatica will run with this Win OS Version -> is there a suitable client?

marcfir, 2023-04-18:
> @marcfir Can you please have a look at this?

Should be fixed with your PR #387 and [58a6886](https://github.com/umati/Dashboard-OPCUA-Client/pull/387/commits/58a688694002f374ef60ee347c799bd76dbc035a). Please test the binaries of your PR in a VM with Windows 32 bit.

marcfir, 2023-04-21:
> Please tell me which versions can be used for: (ideally with a link) WES7V4.2.0 (32 Bit) and which for Windows10IoTV5.1.0 (32 Bit).

@BFrKUKA Your named systems should work with [release-windows-2019-386](https://github.com/umati/Dashboard-OPCUA-Client/suites/12195383034/artifacts/644212595) from PR #387. You can find the artifacts [here](https://github.com/umati/Dashboard-OPCUA-Client/actions/runs/4682397378?pr=387). Make sure that the Visual C++ Redistributable is installed on your Win7 system [see here](https://learn.microsoft.com/de-de/cpp/windows/latest-supported-vc-redist?view=msvc-170). You will need [https://aka.ms/vs/17/release/vc_redist.x86.exe](https://aka.ms/vs/17/release/vc_redist.x86.exe).

I have tested the artifact on Win7SP1 Build 7601. Please let us know if you are successful on your systems with the name artifacts.

Kantiran91, 2023-04-27:
Seems to work now - as the client publish to staging mqtt broker.
@xenonforlife @BFrKUKA: I close this issue

closed 2023-04-27 by Kantiran91