umati/Dashboard-OPCUA-Client #454: [TTD] Import of the DataType not work with example server.

Kantiran91, 2023-05-05:
See Log for more information.
@mdornaus - Maybe this issue helps debugging the Ted example server.

```
2023-05-05 08:44:11,525 [DashboardOpcU

aClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 0, namespaceURI: http://opcfoundation.org/UA/

2023-05-05 08:44:11,525 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 1, namespaceURI: urn:nb-pf3x85f5:UA:Quickstarts:ReferenceServer

2023-05-05 08:44:11,525 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 2, namespaceURI: http://opcfoundation.org/UA/DI/

2023-05-05 08:44:11,526 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 3, namespaceURI: http://opcfoundation.org/UA/Machinery/

2023-05-05 08:44:11,526 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 4, namespaceURI: http://opcfoundation.org/UA/Machinery/Result/

2023-05-05 08:44:11,526 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 5, namespaceURI: http://opcfoundation.org/UA/IA/

2023-05-05 08:44:11,526 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 6, namespaceURI: http://opcfoundation.org/UA/MachineTool/

2023-05-05 08:44:11,526 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 7, namespaceURI: http://opcfoundation.org/UA/TTD/

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 8, namespaceURI: http://opcfoundation.org/UA/TTD/Instance

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 9, namespaceURI: http://opcfoundation.org/UA/TTD/Instance2

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 10, namespaceURI: http://opcfoundation.org/UA/Diagnostics

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 11, namespaceURI: System.String[]

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):627 Namespace already in cache

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 12, namespaceURI: System.String[]

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):627 Namespace already in cache

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 13, namespaceURI: System.String[]

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):627 Namespace already in cache

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 14, namespaceURI: System.String[]

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):627 Namespace already in cache

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 15, namespaceURI: System.String[]

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):627 Namespace already in cache

2023-05-05 08:44:11,527 [DashboardOpcUaClient] INFO void Umati::OpcUa::OpcUaClient::fillNamespaceCache(const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&):630 index: 16, namespaceURI: System.String[]

2023-05-05 08:44:11,532 [DashboardOpcUaClient] ERROR void Umati::OpcUa::Subscription::createSubscription(UA_Client*):101 Create subscription succeeded, id 4

2023-05-05 08:44:11,533 [DashboardOpcUaClient] INFO Umati::MqttPublisher_Paho::MqttPublisher_Paho::MqttPublisher_Paho(const std::string&, const std::string&, uint16_t, const std::string&, const std::string&, const std::string&, const std::string&, const std::string&, const std::string&, const std::string&):50 Connect to 192.168.178.23

2023-05-05 08:44:11,742 [DashboardOpcUaClient] INFO virtual void Umati::MqttPublisher_Paho::MqttPublisher_Paho::MqttCallbacks::connected(const std::string&):125 Mqtt Connected: connect onSuccess called

2023-05-05 08:44:11,782 [DashboardOpcUaClient] INFO virtual void Umati::OpcUa::OpcUaClient::readTypeDictionaries():369 Unable to resolve DataType:http://opcfoundation.org/UA/Machinery/Result/ BaseResultTransferOptionsDataType

2023-05-05 08:44:11,851 [DashboardOpcUaClient] ERROR virtual void Umati::OpcUa::OpcUaClient::buildCustomDataTypes():224 Error occured searching for ua:KeyValuePair: map::at

2023-05-05 08:44:11,851 [DashboardOpcUaClient] ERROR virtual void Umati::OpcUa::OpcUaClient::buildCustomDataTypes():224 Error occured searching for ns1:ProcessingTimesDataType: map::at

2023-05-05 08:44:11,851 [DashboardOpcUaClient] ERROR virtual void Umati::OpcUa::OpcUaClient::buildCustomDataTypes():224 Error occured searching for ns1:ResultEvaluationEnum: map::at

2023-05-05 08:44:11,852 [DashboardOpcUaClient] INFO void Umati::Dashboard::OpcUaTypeReader::readTypes():58 Browsing variable types.

2023-05-05 08:44:12,803 [DashboardOpcUaClient] INFO std::shared_ptr<ModelOpcUa::StructureBiNode> Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(BiDirTypeMap_t&, const ModelOpcUa::BrowseResult_t&, const std::weak_ptr<ModelOpcUa::StructureBiNode>&, ModelOpcUa::ModellingRule_t, bool):319 Current size BiDirectionalTypeMap: 50

2023-05-05 08:44:12,991 [DashboardOpcUaClient] INFO void Umati::Dashboard::OpcUaTypeReader::readTypes():60 Browsing variable types finished, continuing browsing object types

[... 3 lines cut ...]

2023-05-05 08:44:14,384 [DashboardOpcUaClient] INFO std::shared_ptr<ModelOpcUa::StructureBiNode> Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(BiDirTypeMap_t&, const ModelOpcUa::BrowseResult_t&, const std::weak_ptr<ModelOpcUa::StructureBiNode>&, ModelOpcUa::ModellingRule_t, bool):319 Current size BiDirectionalTypeMap: 200

[... cut ...]
2023-05-05 08:44:18,928 [DashboardOpcUaClient] ERROR UA_NodeClass Umati::OpcUa::OpcUaClient::readNodeClass(const open62541Cpp::UA_NodeId&):472 readNodeClass failed

2023-05-05 08:44:18,928 [DashboardOpcUaClient] ERROR UA_NodeClass Umati::OpcUa::OpcUaClient::readNodeClass(const open62541Cpp::UA_NodeId&):476 readNodeClass failed

2023-05-05 08:44:18,928 [DashboardOpcUaClient] ERROR UA_NodeClass Umati::OpcUa::OpcUaClient::nodeClassFromNodeId(const open62541Cpp::UA_NodeId&):756 Invalid NodeClass 0 expect object or variable type for node 

terminate called after throwing an instance of 'Umati::Exceptions::UmatiException'

  what():  Invalid NodeClass
```

Kantiran91, 2023-05-26:
@mdornaus This is fixed? Need we an PR of the itma branch?

GoetzGoerisch, 2025-03-21:
Follow up in umatiGateway if needed.

closed 2025-03-21 by GoetzGoerisch