umati/Dashboard-OPCUA-Client #462: umati Client Crashed when connecting to opcua.umati.app:443

Kantiran91, 2023-05-11:
Seems there is an empty nodeId in the types

```
2023-05-11 12:06:54,382 [DashboardOpcUaClient] ERROR void __thiscall Umati::OpcUa::Converter::UaDataValueToJsonValue::setValueFromScalarVariant(UA_Variant &,class nlohmann::json_abi_v3_11_2::basic_json<class std::map,class std::vector,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,bool,__int64,unsigned __int64,double,class std::allocator,struct nlohmann::json_abi_v3_11_2::adl_serializer,class std::vector<unsigned char,class std::allocator<unsigned char> >,void> *,bool):258 Unknow conversion from OpcUaType_ExtensionObject with DataTypeEncodingType: ns=0; i=0
2023-05-11 12:06:54,383 [DashboardOpcUaClient] ERROR void __thiscall Umati::OpcUa::Converter::UaDataValueToJsonValue::setValueFromScalarVariant(UA_Variant &,class nlohmann::json_abi_v3_11_2::basic_json<class std::map,class std::vector,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,bool,__int64,unsigned __int64,double,class std::allocator,struct nlohmann::json_abi_v3_11_2::adl_serializer,class std::vector<unsigned char,class std::allocator<unsigned char> >,void> *,bool):266 The erroc originates from ns=21; s=55240
[2023-05-11 12:06:54.383 (UTC+0200)] error/eventloop    Cannot run EventLoop from the run method itself
[2023-05-11 12:06:54.384 (UTC+0200)] error/eventloop    Cannot run EventLoop from the run method itself
Segmentation fault
```

Kantiran91, 2023-05-12:
Yes I´m sure its the node-opcua server 
I think it´s the same issue your report on the chat 
Your write the name of the issue is bad.
Version: 2.0.0-rc6-3-ga05ab53-dirty

GoetzGoerisch, 2023-05-12:
@AndreasHeine  any Idea?

GoetzGoerisch, 2023-05-12:
@Kantiran91 currently the client at `2.0.0-rc6-7-g2edc2ef-dirty` is not even able to subscribe to machines, as it is looking for MachineInstances at the TypeDefinitions.

GoetzGoerisch, 2023-05-12:
From where did you build `ga05ab53`?

Kantiran91, 2023-05-12:
the ITMA Branch

GoetzGoerisch, 2023-05-15:
rc3 is partially working with a lot of new error messages.

AndreasHeine, 2023-05-15:
"ns=21; s=55240"

is that a real nodeid?

Kantiran91, 2023-05-15:
@mdornaus Do we have changed something that is relevant to the nodeids?

Kantiran91, 2023-05-15:
New log is:
`2023-05-15 23:01:24,281 [DashboardOpcUaClient] ERROR void Umati::OpcUa::Converter::UaDataValueToJsonValue::setValueFromScalarVariant(UA_Variant &, nlohmann::json *, bool):258 Unknow conversion from OpcUaType_ExtensionObject with DataTypeEncodingType: ns=0; i=0
2023-05-15 23:01:24,281 [DashboardOpcUaClient] ERROR void Umati::OpcUa::Converter::UaDataValueToJsonValue::setValueFromScalarVariant(UA_Variant &, nlohmann::json *, bool):262 The erroc originates from ns=ns=21;i=55230
[2023-05-15 23:01:24.281 (UTC+0200)] error/eventloop	Cannot run EventLoop from the run method itself
2023-05-15 23:01:24,281 [DashboardOpcUaClient] ERROR void Umati::OpcUa::Converter::UaDataValueToJsonValue::setValueFromScalarVariant(UA_Variant &, nlohmann::json *, bool):268 could not read DataType Attribute
libc++abi: terminating with uncaught exception of type Umati::Exceptions::OpcUaNonGoodStatusCodeException: 80020000, BadInternalError: Could not read DataType Attribute`

AndreasHeine, 2023-05-16:
ShowCaseMachineTool.xml:
```xml

<UAVariable DataType="Double" ParentNodeId="ns=1;i=55228" NodeId="ns=1;i=55229" BrowseName="3:FeedOverride">
        <DisplayName>FeedOverride</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=17570</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=55231</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=55230</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=55228</Reference>
        </References>
    </UAVariable>

<!--
...
-->

<UAVariable DataType="Range" ParentNodeId="ns=1;i=55229" NodeId="ns=1;i=55230" BrowseName="EURange">
        <DisplayName>EURange</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=55229</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
</UAVariable>
```

its just an EURange... should be no issue at all (builtin structure!)

Kantiran91, 2023-05-16:
I have tested the different Release Candidate - the rc5 is the problem.
There must be a bug some where in this code part:
https://github.com/umati/Dashboard-OPCUA-Client/blob/664326a29ff1ab367ebaaa2539d128aa26a2ed00/OpcUaClient/Converter/UaDataValueToJsonValue.cpp#L231-L253

PR #470 (2023-05-31): feat(core): add case distraction for extendedobject encoding
The PR fix #462
closed 2023-05-31 by GoetzGoerisch