umati/Dashboard-OPCUA-Client #471: Client cannot resolve InterfaceType correct

Kantiran91, 2023-05-17:
Log:
`2023-05-17 11:09:17,327 [DashboardOpcUaClient] WARNING void Umati::Dashboard::DashboardClient::preparePlaceholderNodesTypeId(const std::shared_ptr<const ModelOpcUa::StructurePlaceholderNode> &, std::shared_ptr<ModelOpcUa::PlaceholderNode> &, std::list<ModelOpcUa::BrowseResult_t> &):395 Could not find a possible type for nsu=. Continuing without a candidate.
2023-05-17 11:09:17,328 [DashboardOpcUaClient] INFO void Umati::Dashboard::DashboardClient::preparePlaceholderNodesTypeId(const std::shared_ptr<const ModelOpcUa::StructurePlaceholderNode> &, std::shared_ptr<ModelOpcUa::PlaceholderNode> &, std::list<ModelOpcUa::BrowseResult_t> &):379 Updated TypeDefinition of ComputingDevice to nsu=o`??
                                                                                                                                            p?o`??
                                                                                                                                                  p??
                                                                                                                                                     pH??
                                                                                                                                                         phB?
                                                                                                                                                             p?`???
                                                                                                                                                                   p>	?pD ??@`0??
                                                                                                                                                                                   p?>?
                                                                                                                                                                                       0;??
                                                                                                                                                                                           p??
                                                                                                                                                                                               because the node implements an interface
2023-05-17 11:09:17,328 [DashboardOpcUaClient] WARNING void Umati::Dashboard::DashboardClient::preparePlaceholderNodesTypeId(const std::shared_ptr<const ModelOpcUa::StructurePlaceholderNode> &, std::shared_ptr<ModelOpcUa::PlaceholderNode> &, std::list<ModelOpcUa::BrowseResult_t> &):395 Could not find a possible type for nsu=. Continuing without a candidate.`

SDK: Unified Automation
Client Version: 2.0.0-rc6-11-g664326a-dirty
Companion Specification: MV Part 2

An example Server can be used.

GoetzGoerisch, 2023-05-26:
Which server is this? The node-opcua sample server with MV2 is not complete.

Kantiran91, 2023-05-26:
It is an example server from @xenonforlife based an Unified Automation

GoetzGoerisch, 2023-05-26:
@xenonforlife how is this different from a instance perspective than the node-opcua server?

closed 2025-03-21 by GoetzGoerisch