umati/Dashboard-OPCUA-Client #485: Machinery Components not showen

Kantiran91, 2023-05-30:
@Kantiran91 

I have tried the latest version of the gateway, now, it is NOT showing the Asset Data in the following umati.app:

https://umati.app/all/TighteningSystemType/clientId%3Datlas%7Cixb%3Bnsu%3Dhttp%3A_2F_2Fwww.atlascopco.com_2FTighteningApplication_2F%3Bs%3D_2FObjectsFolder_2FTighteningSystem_5FAtlasCopco 

[image]

With the older version of the gateway, the asset data is visible in another machine as shown below:

https://umati.app/all/TighteningSystemType/clientId%3Datlas%7Cpf8%3Bnsu%3Dhttp%3A_2F_2Fwww.atlascopco.com_2FTighteningApplication_2F%3Bs%3D_2FObjectsFolder_2FTighteningSystem_5FAtlasCopco

_Originally posted by @mohitkumaragarwal in https://github.com/umati/Dashboard-OPCUA-Client/issues/361#issuecomment-1567228205_

Kantiran91, 2023-05-30:
@GoetzGoerisch 

I looked at the log and found this:

`2023-05-30 10:06:08,744 [DashboardOpcUaClient] INFO class std::list<struct ModelOpcUa::BrowseResult_t,class std::allocator<struct ModelOpcUa::BrowseResult_t> > __thiscall Umati::MachineObserver::MachineObserver::browseForMachines(struct ModelOpcUa::NodeId_t,struct ModelOpcUa::NodeId_t,class std::function<bool __cdecl(struct ModelOpcUa::NodeId_t)>):224 IdentificationType not found, probably because namesapce nsu=http://opcfoundation.org/UA/IJT/;i=1003 is not in the config, or this type does   not have an identification type.`

This are the Type of:

- IControllerType
- IToolType

Could it possible that the components also need to be set in the config?

mohitkumaragarwal, 2023-05-31:
@Kantiran91 I have attached the configuration file which I am using with the umati gateway. 

The same file was used with previous versions and newer versions of the gateway.

It does NOT have the entry for 1003 or 1004.

[configuration.txt](https://github.com/umati/Dashboard-OPCUA-Client/files/11614244/configuration.txt)

Kantiran91, 2023-06-05:
@mdornaus I can locale the issue is since f057dc2e005ce964a5402cda7b40996c0c581389 - Can you please look at this.

mdornaus, 2023-06-07:
@ccvca @wlkrm Ok I tracked the issue down and it seems that someting is wrong with the void OpcUaTypeReader::updateTypeMap() method in the TypeReader. I don't exactly understand what this method is supposed to do especialy with the folders. It seems that everything works fine if I put the method out. So maybe you could explain me why this method is needed. (Especially the folder part).

Kantiran91, 2023-06-07:
> This method ensures, that all optional members of a type are searched, event if the optional nodes are only defined in the type of the child and not the child itself.

Than i assume that we need to build this recursive and the current implementation works only in the depth of two and the spec has defined in the depth of three.

closed 2023-07-19 by GoetzGoerisch