umati/Dashboard-OPCUA-Client #490: Add support for linux arm 64

izonfreak, 2023-06-01:
For software that runs inside embedded devices like the jetson nvidia nano or if you want to run the gateway in an embedded device like a raspberry-pi. It will be nice to have a binary ready instead of having to figure out how to compile the project for arm.

wlkrm, 2023-06-06:
Have you figured out how to compile the project for arm? Could you perhaps create a pull request or post the necessary changes here?

GoetzGoerisch, 2023-07-20:
Do you want a standalone binary or just a container image for `arm64`?

GoetzGoerisch, 2023-07-28:
Container image to try: ghcr.io/umati/dashboard-opcua-client:feat-arm-build
Standalone linux binary:

GoetzGoerisch, 2025-03-21:
We are switching to umatiGateway for the future. @izonfreak please try the binaries here: https://github.com/umati/umatiGateway/actions/runs/13971748770

closed 2025-03-21 by GoetzGoerisch