umati/Dashboard-OPCUA-Client #495: Dashboard Client in Container cannot connect to MQTT

HaraldRentschler, 2023-06-02:
### Command:
**docker run -it --rm -v ./Configuration.json:/app/configuration.json --name=dashboard-opcua-client ghcr.io/umati/dashboard-opcua-client:release-2.0.0**

Connection to OPC-UA Server works as expected, but there is always an PAHO exception. What is wrong? Configuration.json works with Dashboard Client started as a normal program not in a container. 

### Output:
2023-06-02 13:14:08,967 [DashboardOpcUaClient] INFO void Umati::Util::ConfigureLogger(const std::string&):44 Logger Configured: DashboardOpcUaClient
2023-06-02 13:14:08,967 [DashboardOpcUaClient] INFO int main(int, char**):62 Start Dashboard OPC UA Client
2023-06-02 13:14:08,967 [DashboardOpcUaClient] INFO int main(int, char**):63 OPC UA Dashboard Client Version: 2.0.0-rc7-dirty

2023-06-02 13:14:08,967 [DashboardOpcUaClient] WARNING Umati::OpcUa::Subscription::Subscription(const std::map<std::__cxx11::basic_string<char>, short unsigned int>&, const std::map<short unsigned int, std::__cxx11::basic_string<char> >&):53 Created subscription 0x7f14b610b968
[2023-06-02 13:14:08.967 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
2023-06-02 13:14:08,967 [DashboardOpcUaClient] INFO static void Umati::OpcUa::SetupSecurity::createNewClientCert():192 Creating new client certificate
[2023-06-02 13:14:08.967 (UTC+0000)] info/channel       Create Certificate: Generating RSA key. This may take a while.
[2023-06-02 13:14:09.037 (UTC+0000)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2023-06-02 13:14:09.037 (UTC+0000)] info/securitypolicy        The basic256 security policy with openssl is added.
[2023-06-02 13:14:09.038 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2023-06-02 13:14:09.042 (UTC+0000)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2023-06-02 13:14:09.042 (UTC+0000)] info/eventloop     Starting the EventLoop
[2023-06-02 13:14:09.042 (UTC+0000)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-06-02 13:14:09.043 (UTC+0000)] info/network       TCP 6   | New connection to "10.1.29.30" on port 48010
2023-06-02 13:14:09,043 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):39 Client is connecting
2023-06-02 13:14:09,043 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
2023-06-02 13:14:09,045 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):45 Client hello sent
2023-06-02 13:14:09,045 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
2023-06-02 13:14:09,047 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):57 Client open sent
2023-06-02 13:14:09,047 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
[2023-06-02 13:14:09.048 (UTC+0000)] info/channel       TCP 6   | SC 124598819  | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-06-02 13:14:09.048 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
2023-06-02 13:14:09,048 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):36 A SecureChannel to the server is open
2023-06-02 13:14:09,048 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
[2023-06-02 13:14:09.052 (UTC+0000)] info/client        Use the EndpointURL opc.tcp://opcuaserver:48010 returned from FindServers
2023-06-02 13:14:09,053 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):60 Client closing
2023-06-02 13:14:09,053 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
[2023-06-02 13:14:09.053 (UTC+0000)] info/channel       TCP 0   | SC 0  | Closed the SecureChannel
[2023-06-02 13:14:09.053 (UTC+0000)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-06-02 13:14:09.053 (UTC+0000)] info/network       TCP 7   | New connection to "10.1.29.30" on port 48010
2023-06-02 13:14:09,053 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):39 Client is connecting
2023-06-02 13:14:09,053 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
[2023-06-02 13:14:09.053 (UTC+0000)] info/network       TCP 6   | Socket closed
2023-06-02 13:14:09,063 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):45 Client hello sent
2023-06-02 13:14:09,063 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
2023-06-02 13:14:09,065 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):57 Client open sent
2023-06-02 13:14:09,065 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
[2023-06-02 13:14:09.066 (UTC+0000)] info/channel       TCP 7   | SC 124598820  | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-06-02 13:14:09.067 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
2023-06-02 13:14:09,067 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):36 A SecureChannel to the server is open
2023-06-02 13:14:09,067 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):72 Session disconnected
[2023-06-02 13:14:09.069 (UTC+0000)] info/client        Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
[2023-06-02 13:14:09.069 (UTC+0000)] info/client        Selected endpoint 0 in URL opc.tcp://opcuaserver:48010 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-06-02 13:14:09.069 (UTC+0000)] info/client        Selected UserTokenPolicy UserName-[0]-None-None with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
2023-06-02 13:14:09,069 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):36 A SecureChannel to the server is open
2023-06-02 13:14:09,069 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):75 Session create requested
[2023-06-02 13:14:09.072 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy channel with openssl is created.
[2023-06-02 13:14:09.072 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy channel with openssl is deleted.
[2023-06-02 13:14:09.072 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
2023-06-02 13:14:09,072 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):36 A SecureChannel to the server is open
2023-06-02 13:14:09,072 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):78 Session created
[... cut ...]
[2023-06-02 13:14:28.425 (UTC+0000)] info/network       TCP 7   | Socket closed
[2023-06-02 13:14:28.524 (UTC+0000)] info/eventloop     Stopping the EventLoop
[2023-06-02 13:14:28.524 (UTC+0000)] info/network       UDP     | Shutting down the ConnectionManager
[2023-06-02 13:14:28.524 (UTC+0000)] info/network       TCP     | Shutting down the ConnectionManager
[2023-06-02 13:14:28.524 (UTC+0000)] info/eventloop     The EventLoop has stopped
2023-06-02 13:14:28,524 [DashboardOpcUaClient] INFO int main(int, char**):109 End Dashboard OPC UA Client

GoetzGoerisch, 2023-06-08:
@HaraldRentschler just to be clear, the same config on the same host works with standalone binary?

Could you please run the WSSTroubleshooter.exe with this configuration.json and report the results?

Please try the correct RC7 container tag: ghcr.io/umati/dashboard-opcua-client:`v2.0.0-rc7`

Thank you.

HaraldRentschler, 2023-06-12:
I used the same config on the same host. I tried different version of the dashboard container always with the same result.  Also the v2.0.0-rc7. Meanwhile I did some additional test.  I think there is a problem with different docker versions and the dashboard containers. When I start the docker command wirh sudo it works. On this VM docker version 24.0.2 is used. On an other VM with docker 20.x.x. it works without sudo rights. But on this VM the standalone client works only with sudo rights. 

Output WssTroubleshooter:

/app # ./WssTroubleshooter 
Connecting to the MQTT server wss://staging.umati.app:443/ws ... Connection attempt failed
Connection attempt failed
Connection attempt failed
Connection attempt failed

GoetzGoerisch, 2023-06-12:
Thank you @HaraldRentschler.

The requirement for sudo, depends on the Docker installation, if it is rootless or not.
The debian VM we are using for production, requires sudo for the docker command.

Regarding the WSSTroubleshooter, there seems to be an issue with a TLS inspection firewall if I'm not mistake, as you are not able to connect to the MQTT broker via WebSockets.

From this network, did you try to reach the MQTT broker via WSS with a GUI client like MQTT Explorer or such?
Easy check is also with a webbrowser, if you visit the staging.umati.app and check for the websites TLS cert as discribed here: 
https://github.com/umati/Dashboard-OPCUA-Client/blob/development/doc/Troubleshooting.md#using-the-gateway-with-firewall-and-tls-inspection

HaraldRentschler, 2023-06-12:
docker runs without sudo, but the dashboard client get no connection with the paho mqtt client. The connection to the opc ua server works. Why do the paho mqtt client need sudo rights to work?

wlkrm, 2023-07-04:
Hello @HaraldRentschler, could you please provide some information of which operating system is running on your VM (the one, where the connection does not work without root?). There might be many causes for that, e.g. AppArmor, SELinux, etc. requiring whitelisting of applications, which are allowed to do some network specific configuration. For example setsockopt might be not allowed for non-root applications.

wlkrm, 2023-07-04:
@HaraldRentschler can you run "ping umati.app" from your vm's commandline without sudo and with sudo? Is there any difference?

wlkrm, 2023-07-04:
I think it could be related to DNS resolving issues, as the connection works for the OPC UA Server, which probably is a plain IP address. Here someone is describing a similar issue for his VM setup: 
https://stackoverflow.com/questions/49998099/dns-not-working-within-docker-containers-when-host-uses-dnsmasq-and-googles-dns

You could try running the container with host network.
```
docker run -it --rm --network host -v ./Configuration.json:/app/configuration.json --name=dashboard-opcua-client ghcr.io/umati/dashboard-opcua-client:release-2.0.0
```

wlkrm, 2023-07-04:
You can verify, if the issue is related to DNS:

1. Run `sudo docker run --rm -it arunvelsriram/utils bash` to start container with command line access
2. Run `nslookup umati.app`. Output  should look like this:
```
utils@142d7a43a461:~$ nslookup umati.app
Server:		129.69.252.252
Address:	129.69.252.252#53

Non-authoritative answer:
Name:	umati.app
Address: 34.117.226.167
Name:	umati.app
Address: 2600:1901:0:8ede::
```
3. Run  `docker run --rm -it arunvelsriram/utils bash` to start container with command line access
4. Run `nslookup umati.app`. Output  should look like this:
```
utils@142d7a43a461:~$ nslookup umati.app
Server:		129.69.252.252
Address:	129.69.252.252#53

Non-authoritative answer:
Name:	umati.app
Address: 34.117.226.167
Name:	umati.app
Address: 2600:1901:0:8ede::
```
5. Run  `docker run --network host --rm -it arunvelsriram/utils bash` to start container with command line access
6. Run `nslookup umati.app`. Output  should look like this:
```
utils@142d7a43a461:~$ nslookup umati.app
Server:		129.69.252.252
Address:	129.69.252.252#53

Non-authoritative answer:
Name:	umati.app
Address: 34.117.226.167
Name:	umati.app
Address: 2600:1901:0:8ede::
```

If the output looks something like that, we have a DNS issue:
```
utils@63a9ffe8a55f:~$ nslookup umati.app
;; connection timed out; no servers could be reached

utils@63a9ffe8a55f:~$ nslookup umati.app
Server:		129.69.252.252
Address:	129.69.252.252#53

** server can't find hih: NXDOMAIN

```

HaraldRentschler, 2023-07-24:
https://github.com/umati/Dashboard-OPCUA-Client/issues/495#issuecomment-1619523917 
There is no difference if I run ping with or without sudo.

PING umati.app (34.117.226.167) 56(84) bytes of data.
64 bytes from 167.226.117.34.bc.googleusercontent.com (34.117.226.167): icmp_seq=1 ttl=113 time=13.6 ms
64 bytes from 167.226.117.34.bc.googleusercontent.com (34.117.226.167): icmp_seq=2 ttl=113 time=12.0 ms
64 bytes from 167.226.117.34.bc.googleusercontent.com (34.117.226.167): icmp_seq=3 ttl=113 time=12.7 ms

HaraldRentschler, 2023-07-24:
https://github.com/umati/Dashboard-OPCUA-Client/issues/495#issuecomment-1619538105 
The result is always:
Server:         10.1.6.68
Address:        10.1.6.68#53

Non-authoritative answer:
Name:   umati.app
Address: 34.117.226.167
Name:   umati.app
Address: 2600:1901:0:8ede::

GoetzGoerisch, 2023-07-24:
Great, means that the nameresolution is working.
Could you please try the recently released v2.0.0

If this does not help, please setup a debug call with @wlkrm.

HaraldRentschler, 2023-07-24:
With the new v2.0.0 the exception does not exist anymore.

HaraldRentschler, 2023-07-26:
The new release has the same problem.

wlkrm, 2023-07-27:
I still cannot reproduce this, even with truly rootless docker (https://docs.docker.com/engine/security/rootless/) or just running docker without root. Even using another user who is not in the sudoers group (tested on debian10):
```
docker run --user 1000:1000 --name client -it --rm -v /home/debian/configuration.json:/app/configuration.json --entrypoint /bin/sh ghcr.io/umati/dashboard-opcua-client:development
```

wlkrm, 2023-07-28:
Solved now, the issue is some necessary setting to use virtualbox in a company network. For future users with the same issue, see @HaraldRentschler 's findings below:

"it is now clear which setting plays a role for the user in VirtualBox. In the ~/.docker/config.json proxies were defined. These were necessary for me, because of company settings in the network. Why here the getaddrinfo goes wrong, local proxy with this address ran on my Windows computer, is not yet clear, but this is the reason why the connection was not established."

wlkrm, 2023-07-28:
Thanks @HaraldRentschler for your information, we will add options for verbose logging in the next release.

wlkrm, 2023-07-28:
@HaraldRentschler, if you find a solution how to setup the proxy correctly, please consider adding the information here, as other users might run into similar issues.

closed 2023-07-28 by wlkrm