umati/Dashboard-OPCUA-Client #500: Dashboard page for machine not displaying.

izonfreak, 2023-06-06:
It seems that I have broken the umati.app dashboard. 

I created a subtype of my machine type to add a variable to it, before I was just using the standard one. Once I added this variable and changed the configuration.json the dashboard from my machine is no longer displaying. You can find the machine if you go to the virtual fair but everything is blank. 

https://umati.app/all/VVisionSystemAssetType/clientId%3Dmvtec%7Cautomatica1%3Bnsu%3Durn%3Amesse-3090%3AMVTec%3AMachineVisionServer%3Bs%3D1.VisionSystemAsset

Kantiran91, 2023-06-06:
@izonfreak It would help us if you can send the additional variable and a screenshot of the type definition and the instance.

izonfreak, 2023-06-07:
The additional variable is of the ResultMetaDataType from machinery. Additionally, this sub-type adds a ResultManagmentType object as an optional component to the type.  This 

When I run the gateway locally the json seem to be working as expected, it was picking the ResultMetadata variable. But as soon as I connected it to umati.app my dashboard disappear. 

The variable exist because issues #501 and #497 block me from doing anything else that will be more correct according to the standards.

Kantiran91, 2023-06-07:
Yes because the Dashboard do not know this Type an has no mapping to the UI elements.