umati/Dashboard-OPCUA-Client #535: [Woodworking] Gateway does not subscribe to Woodworking Types

GoetzGoerisch, 2023-06-27:
**Describe the bug**
Using RC7 does not work with Woodworking CS servers.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect Gateway RC7 with opc.tcp://opcua.umati.app:4843
2. Wait for subscription of the Woodworking instances
3. Observe published information in MQTT broker

**Expected behavior**
Expected output:
```json
{
  "Identification": {
    "DeviceClass": "Class A12",
    "Manufacturer": {
      "locale": "",
      "text": ""
    },
    "Model": {
      "locale": "",
      "text": ""
    },
    "ProductInstanceUri": "https://www.example.com",
    "SerialNumber": "378485745354392",
    "YearOfConstruction": 2023
  },
  "State": {
    "Machine": {
      "Flags": {
        "Alarm": false,
        "Calibrated": false,
        "Emergency": true,
        "Error": false,
        "MachineInitialized": false,
        "MachineOn": false,
        "PowerPresent": false,
        "RecipeInRun": false,
        "Warning": true
      },
      "Overview": {
        "CurrentMode": 5,
        "CurrentState": 4
      }
    }
  }
}
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment:**

- OS: Linux Container
- OPC UA Dashboard Client Version [RC7]
- Used Enviroment [umati.app]
- OPC UA SDK of server [node-opcua]

GoetzGoerisch, 2023-06-27:
RC 4 works.

Kantiran91, 2023-06-27:
@mdornaus If I remember correct, woodworking also use interface. Could be related to our problem with the interface two weeks ago?

GoetzGoerisch, 2023-07-21:
#559 did not fix this issue.

closed 2023-07-21 by GoetzGoerisch