umati/Dashboard-OPCUA-Client #557: DashboardOpcUaClient - Unable to find namespace

RitterDerKokosnuss, 2023-07-17:
Hi,

i recently started working with OpcUa.

Currently i try to use the umati DashboardOpcUaClient, but get the following errors when running:

`- Type not found. Probably because the server does not have the NodeId, specified in the config (nsu=https://opcfoundation.org/UA/MachineTool/;i=13)`

```
2023-07-17 11:04:15,328 [DashboardOpcUaClient] INFO void __cdecl Umati::Dashboard::OpcUaTypeReader::readTypes(void):63 Browsing object types finished
2023-07-17 11:04:15,328 [DashboardOpcUaClient] WARNING void __cdecl Umati::Dashboard::OpcUaTypeReader::readTypes(void):85 Unable to find namespace http://opcfoundation.org/UA/IA/
2023-07-17 11:04:15,328 [DashboardOpcUaClient] WARNING void __cdecl Umati::Dashboard::OpcUaTypeReader::readTypes(void):85 Unable to find namespace http://opcfoundation.org/UA/DI/
2023-07-17 11:04:15,328 [DashboardOpcUaClient] WARNING void __cdecl Umati::Dashboard::OpcUaTypeReader::readTypes(void):85 Unable to find namespace http://opcfoundation.org/UA/Machinery/
2023-07-17 11:04:15,328 [DashboardOpcUaClient] WARNING void __cdecl Umati::Dashboard::OpcUaTypeReader::readTypes(void):85 Unable to find namespace http://opcfoundation.org/UA/MachineTool/
2023-07-17 11:04:15,344 [DashboardOpcUaClient] ERROR class std::shared_ptr<class ModelOpcUa::StructureNode> __cdecl Umati::Dashboard::OpcUaTypeReader::typeDefinitionToStructureNode(const struct ModelOpcUa::NodeId_t &) const:363 Unable to find nsu=http://opcfoundation.org/UA/MachineTool/;i=13 in typeMap
2023-07-17 11:04:15,344 [DashboardOpcUaClient] WARNING void __cdecl Umati::Dashboard::OpcUaTypeReader::updateObjectTypeNames(void):104 Type not found. Probably because the server does not have the NodeId, specified in the config (nsu=http://opcfoundation.org/UA/MachineTool/;i=13).
2023-07-17 11:04:15,344 [DashboardOpcUaClient] INFO bool __cdecl Umati::MachineObserver::MachineObserver::canBrowseMachineList(class std::list<struct ModelOpcUa::BrowseResult_t,class std::allocator<struct ModelOpcUa::BrowseResult_t> > &):129 Searching for machines
2023-07-17 11:04:15,356 [DashboardOpcUaClient] ERROR unsigned short __cdecl Umati::OpcUa::Converter::ModelToUaConverter::getNsIndexFromUri(const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &):29 Could not find uri: http://opcfoundation.org/UA/Machinery/
```

When i use UAExpert and NodeRED (OPCUA IIot) i have access to all nodes and can read, write and subscribe without any issues. Only with the DashboardOpcUaClient i run into this error.

I tried changing the configuration.json file but without success.

Am i missing a step or some dependency?

Any help will be appreciated

Kantiran91, 2023-07-18:
@RitterDerKokosnuss Which SDK do you use? and what do you mean that your server not listed the cs namespaces? The Dashboard Client need the namespaces.

Can you send us a screenshot of the UAExpert?

RitterDerKokosnuss, 2023-07-18:
[image]

RitterDerKokosnuss, 2023-07-19:
Today the OpcUaClient produced the following output as long as the machine was running. After restarting the machine, old error is showing again.

```
std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &):50 Connect to umati.app
2023-07-19 10:35:14,622 [DashboardOpcUaClient] INFO void __cdecl Umati::MqttPublisher_Paho::MqttPublisher_Paho::MqttCallbacks::connected(const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &):125 Mqtt Connected: connect onSuccess called
2023-07-19 10:35:14,622 [DashboardOpcUaClient] INFO void __cdecl Umati::Dashboard::OpcUaTypeReader::readTypes(void):58 Browsing variable types.
2023-07-19 10:35:15,518 [DashboardOpcUaClient] INFO class std::shared_ptr<class ModelOpcUa::StructureBiNode> __cdecl Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(class std::shared_ptr<class std::map<struct ModelOpcUa::NodeId_t,class std::shared_ptr<class ModelOpcUa::StructureBiNode>,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct std::pair<struct ModelOpcUa::NodeId_t const ,class std::shared_ptr<class ModelOpcUa::StructureBiNode> > > > > &,const struct ModelOpcUa::BrowseResult_t &,const class std::weak_ptr<class ModelOpcUa::StructureBiNode> &,enum ModelOpcUa::ModellingRule_t,bool):319 Current size BiDirectionalTypeMap: 50
2023-07-19 10:35:16,054 [DashboardOpcUaClient] INFO class std::shared_ptr<class ModelOpcUa::StructureBiNode> __cdecl Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(class std::shared_ptr<class std::map<struct ModelOpcUa::NodeId_t,class std::shared_ptr<class ModelOpcUa::StructureBiNode>,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct std::pair<struct ModelOpcUa::NodeId_t const ,class std::shared_ptr<class ModelOpcUa::StructureBiNode> > > > > &,const struct ModelOpcUa::BrowseResult_t &,const class std::weak_ptr<class ModelOpcUa::StructureBiNode> &,enum ModelOpcUa::ModellingRule_t,bool):319 Current size BiDirectionalTypeMap: 50
2023-07-19 10:35:16,116 [DashboardOpcUaClient] INFO void __cdecl Umati::Dashboard::OpcUaTypeReader::readTypes(void):60 Browsing variable types finished, continuing browsing object types
2023-07-19 10:35:16,431 [DashboardOpcUaClient] INFO class std::shared_ptr<class ModelOpcUa::StructureBiNode> __cdecl Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(class std::shared_ptr<class std::map<struct ModelOpcUa::NodeId_t,class std::shared_ptr<class ModelOpcUa::StructureBiNode>,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct std::pair<struct ModelOpcUa::NodeId_t const ,class std::shared_ptr<class ModelOpcUa::StructureBiNode> > > > > &,const struct ModelOpcUa::BrowseResult_t &,const class std::weak_ptr<class ModelOpcUa::StructureBiNode> &,enum ModelOpcUa::ModellingRule_t,bool):319 Current size BiDirectionalTypeMap: 100
2023-07-19 10:35:16,839 [DashboardOpcUaClient] INFO class std::shared_ptr<class ModelOpcUa::StructureBiNode> __cdecl Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(class std::shared_ptr<class std::map<struct ModelOpcUa::NodeId_t,class std::shared_ptr<class ModelOpcUa::StructureBiNode>,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct std::pair<struct ModelOpcUa::NodeId_t const ,class std::shared_ptr<class ModelOpcUa::StructureBiNode> > > > > &,const struct ModelOpcUa::BrowseResult_t &,const class std::weak_ptr<class ModelOpcUa::StructureBiNode> &,enum ModelOpcUa::ModellingRule_t,bool):319 Current size BiDirectionalTypeMap: 150
2023-07-19 10:35:17,688 [DashboardOpcUaClient] INFO class std::shared_ptr<class ModelOpcUa::StructureBiNode> __cdecl Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(class std::shared_ptr<class std::map<struct ModelOpcUa::NodeId_t,class std::shared_ptr<class ModelOpcUa::StructureBiNode>,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct std::pair<struct ModelOpcUa::NodeId_t const ,class std::shared_ptr<class ModelOpcUa::StructureBiNode> > > > > &,const struct ModelOpcUa::BrowseResult_t &,const class std::weak_ptr<class ModelOpcUa::StructureBiNode> &,enum ModelOpcUa::ModellingRule_t,bool):319 Current size BiDirectionalTypeMap: 200
2023-07-19 10:35:19,619 [DashboardOpcUaClient] INFO class std::shared_ptr<class ModelOpcUa::StructureBiNode> __cdecl Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(class std::shared_ptr<class std::map<struct ModelOpcUa::NodeId_t,class std::shared_ptr<class ModelOpcUa::StructureBiNode>,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct std::pair<struct ModelOpcUa::NodeId_t const ,class std::shared_ptr<class ModelOpcUa::StructureBiNode> > > > > &,const struct ModelOpcUa::BrowseResult_t &,const class std::weak_ptr<class ModelOpcUa::StructureBiNode> &,enum ModelOpcUa::ModellingRule_t,bool):319 Current size BiDirectionalTypeMap: 250
2023-07-19 10:35:21,593 [DashboardOpcUaClient] INFO class std::shared_ptr<class ModelOpcUa::StructureBiNode> __cdecl Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(class std::shared_ptr<class std::map<struct ModelOpcUa::NodeId_t,class std::shared_ptr<class ModelOpcUa::StructureBiNode>,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct std::pair<struct ModelOpcUa::NodeId_t const ,class std::shared_ptr<class ModelOpcUa::StructureBiNode> > > > > &,const struct ModelOpcUa::BrowseResult_t &,const class std::weak_ptr<class ModelOpcUa::StructureBiNode> &,enum ModelOpcUa::Modellin [... cut ...] cUa::NodeId_t>,class std::allocator<struct ModelOpcUa::NodeId_t> > &):283 New / Staying machines: 

2023-07-19 10:35:22,585 [DashboardOpcUaClient] INFO void __cdecl Umati::MachineObserver::MachineObserver::logMachinesChanging(const class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,const class std::set<struct ModelOpcUa::NodeId_t,struct std::less<struct ModelOpcUa::NodeId_t>,class std::allocator<struct ModelOpcUa::NodeId_t> > &):283 Removing machines:  
```

Kantiran91, 2023-07-19:
Hi, 

I think you do not implement the cs for machine tools.  So the client can not find the correct nodes.

RitterDerKokosnuss, 2023-07-20:
Yes and i dont know why. I did everything as described in the siemens documentation. While creating my information model i included /UA-, /IA-, /Machine-, /MachineTools- NodeSet with relevant release versions. Any ideas what i am missing or doing wrong? I use SIOME for information modelling.

GoetzGoerisch, 2023-07-20:
Which Control is this the SINUMERIK integrated OPC UA server is not capable of using companion specifications as far as I'm aware. Please reach out to your Siemens Support.

CC: @AthaDer @HolgnneR

RitterDerKokosnuss, 2023-07-20:
I will reach out to you via mail

closed 2023-08-14 by RitterDerKokosnuss