umati/Dashboard-OPCUA-Client #657: [WW]  Axis Override Values Not Displayed on umati.app

Kantiran91, 2023-12-04:
**Describe the bug**
The Axis Override values are not being displayed on the website. You can view an example of the issue here:

 

[https://url.avanan.click/v2/___https://umati.app/all/WwMachineType/clientId%3Dbiesse%7CSizingTest%3Bnsu%3Dhttp%3A_2F_2Fbiessegroup.com_2FUA_2FWwMachine_2FConfiguration_5FbOPCUAServer_2FMono_2F%3Bs%3DMono___.YXAxZTp2ZG1hOmE6bzo4OTllMzMyYWE1ZGFiYjZiOTM3MjIwZmI1N2Y2ZmZmYjo2OjcxOGU6Mjg1Y2ZjZmYyMGJjYWI5NmY0MTlkMDBmZjBkMzZmOWFjYmU3MDBkNGQzYzAyMmIzOGE5ZTdiMjIzNDY2YmY3NTp0OlQ](https://url.avanan.click/v2/___https:/umati.app/all/WwMachineType/clientId%3Dbiesse%7CSizingTest%3Bnsu%3Dhttp%3A_2F_2Fbiessegroup.com_2FUA_2FWwMachine_2FConfiguration_5FbOPCUAServer_2FMono_2F%3Bs%3DMono___.YXAxZTp2ZG1hOmE6bzo4OTllMzMyYWE1ZGFiYjZiOTM3MjIwZmI1N2Y2ZmZmYjo2OjcxOGU6Mjg1Y2ZjZmYyMGJjYWI5NmY0MTlkMDBmZjBkMzZmOWFjYmU3MDBkNGQzYzAyMmIzOGE5ZTdiMjIzNDY2YmY3NTp0OlQ)

 

On the server side, we are exposing the variable as shown in the attached image.

**To Reproduce**

**Expected behavior**
A clear and concise description of what you expected to happen.
[image]
**Screenshots**

[image]

**Environment:**

- OS: [?]
- OPC UA Dashboard Client Version 2.0.0-rc4 and 2.0.0.-main
- Used Enviroment [umati.app,]
- OPC UA SDK of server [?]

**Log**

**Additional context**

mdornaus, 2024-01-29:
Works with the latest developement-release and the sampleserver. Maybe an Issue with the Machine

[image]

GoetzGoerisch, 2025-03-21:
Should be fixed. Otherwise switch to umatiGateway.

closed 2025-03-21 by GoetzGoerisch