umati/Dashboard-OPCUA-Client #698: Client not working with UaServer (.Net)

GoetzGoerisch, 2024-02-13:
**Describe the bug**
Currently the connection with Ua Server (.Net) from a umati member is not working.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect to a UaServer with the string NodeId

**Expected behavior**
Connection works and is stable

**Environment:**

- OS: Windows 10 
- OPC UA Dashboard Client Version [branch, version]
- Used Enviroment [umati.app]
- OPC UA SDK of server [Ua .NET]

**Log**
Upload log file or relevant error logs.

**Additional context**
Add any other context about the problem here.

GoetzGoerisch, 2024-02-13:
Redacted Log attached

[ClientLog_redacted.txt](https://github.com/umati/Dashboard-OPCUA-Client/files/14259751/ClientLog_redacted.txt)

wlkrm, 2025-02-14:
I think [IJT](https://github.com/umati/UA-for-Industrial-Joining-Technologies/tree/main/OPC_UA_Servers/Release2) is using it and has string NodeIds and I could not reproduce the bug with the IJT-Server.
I noticed that I can only connect to this server running on the same machine using the Ethernet (Default Switch)s IP. localhost, hostname, external IP, ... is not working. Maybe the issue is the same?
`"Endpoint": "opc.tcp://localhost:40451"` does not work but `"Endpoint": "opc.tcp://172.21.240.1:40451"` does.

@GoetzGoerisch, as IJT is now fully integrated could you please trigger them to try the revert/431 branch?

GoetzGoerisch, 2025-03-04:
Make umatiGateway release and migrate user to new gateway.

GoetzGoerisch, 2025-03-21:
Should be improved with latest version. Otherwise use umatiGateway.

closed 2025-03-21 by GoetzGoerisch