umati/Dashboard-OPCUA-Client #747: Error with reduant BrowePath

Kantiran91, 2024-04-18:
**Describe the bug**
If a system has redudant BrowsePath like here:
[image]

The Client raise an error an import not  all data.

**To Reproduce**
use a IJT 2.0 Server 

**Expected behavior**

only use the correct BrowsePath

**Environment:**

- OS: [Container]
- OPC UA Dashboard Client Version [development, ]
- Used Enviroment [dev.umati.app]
- OPC UA SDK of server [ua?]

**Log**

**Additional context**

wlkrm, 2025-02-12:
Redundant browse path is resolved, however does not really help

wlkrm, 2025-02-12:
Could resolve the issue for AssetManagement, it was an error in the server.
The wrong nodes include TighteningSyste->MachineryBuildingBlocks (Should be FolderType ACC. to Spec).
Some additional things seem to be broken as well in the server.

closed 2025-02-12 by wlkrm