umati/Dashboard-OPCUA-Client #811: BadNoMatch during Startup

ChrisDahmen, 2024-07-31:
**Describe the bug**
I cannot use the Client to connect to the Umati Dashboard, upon attempting to start an exception is shown after which the software immediately closes.

**To Reproduce**
Unknown since I cannot not replicate the error

**Expected behavior**
The Application to launch and establish a connection

**Environment:**

- OS: Windows 10 running Mosquitto 2.0.18 as my test Broker, WSL with Docker for the client
- Version Used: Docker Image ghcr.io/umati/dashboard-opcua-client  Tag: renovate-docker-setup-buildx-action-3.x (latest for me) and v2.1.0
- Used Enviroment: Local mosquitto server

- OPC UA SDK of server: Träger's Opc.UaFx.Advanced Version 2.42.1

**Log**
[logfile.txt](https://github.com/user-attachments/files/16440934/logfile.txt)

ainglessi, 2024-08-19:
Hello! The gateway is known to work (with occasional crashes) with a server based on Traeger SDK. Could you please try running the gateway with `"Security": 1` in `"OpcUa"` section of configuration.json and default security settings in the server code with ACL enabled:

```c#
opcServer.Security.AnonymousAcl.IsEnabled = false;
var serverAcl = opcServer.Security.UserNameAcl;
serverAcl.AddEntry("xxx", "xxx");
serverAcl.IsEnabled = true;
```

ChrisDahmen, 2024-08-21:
Hi, I have tried the configuration as given by you, however it still results in the same error. If you need any further files, such as the configuration, to be able to better diagnose the issue please feel free to ask.

Kantiran91, 2024-08-26:
After a test with the old version v2.0.0-rc4 we noticed that some children are missing.

ChrisDahmen, 2024-08-28:
Issue is fixed, now that #816 was found I was able to diagnose the issue.

closed 2024-08-28 by ChrisDahmen