umati/Dashboard-OPCUA-Client #812: Docker Pull not working

ChrisDahmen, 2024-07-31:
**Describe the bug**
Using the command given in the documentation I attempted to pull the docker container, but was faced with an error message instead: 

**To Reproduce**
Use the command from the documentation (docker pull ghcr.io/umati/dashboard-opcua-client)

**Expected behavior**
The Docker container to pulled properly and be runnable

**Environment:**
- OS: Ubuntu WSL container

**Log**
> docker pull ghcr.io/umati/dashboard-opcua-client
> Using default tag: latest
> Error response from daemon: manifest unknown

GoetzGoerisch, 2024-08-19:
We do not tag `latest`, please use `development` to pull the latest development image.

ChrisDahmen, 2024-08-22:
Works using Tag development. A change in documentation ([here](https://github.com/umati/Dashboard-OPCUA-Client/blob/development/doc/Container.md)) might be useful to other newcomers.

closed 2024-08-22 by ChrisDahmen