umati/Dashboard-OPCUA-Client #868: "The server certificate is different from the EndpointDescription" errors and crashes

ainglessi, 2024-12-06:
**Describe the bug**
Starting 22 November the gateway crashes roughly every 15 minutes when connected to the Sample Server (open62541) with 'Umati::Exceptions::ClientNotConnected' error (see logs below).

**To Reproduce**
Steps to reproduce the behavior:

1. Use Sample Server container image version [sha256:65f65cf51b4bc8269625d6d503be31b9e50e8365fbe7e060a3fc85ff6e6636f3](https://github.com/umati/Sample-Server/pkgs/container/sample-server/307749142?tag=renovate-super-linter-super-linter-7.x)
2. Use gateway image version [sha256:31445c2a51904c00af976d98f00b45aded319a8a8ad6f63abf49d5073a0ea4b9](https://github.com/orgs/umati/packages/container/dashboard-opcua-client/310275504?tag=gh-readonly-queue-development-pr-850-ac97bd36f5a2117102242d24e5e1d700dec74caa) (commit https://github.com/umati/Dashboard-OPCUA-Client/commit/ac97bd36f5a2117102242d24e5e1d700dec74caa) or newer.
3. See error in logs and gateway crashing

**Expected behavior**
Works fine with gateway image [sha256:bc655b69bbab29beeb16d25540d42370707104ffcbfcd9e4861fa548f883707b](https://github.com/orgs/umati/packages/container/dashboard-opcua-client/309322909?tag=renovate-deps-googletest-digest) and older.

**Screenshots**
[image]
[image]

**Environment:**

Public `sample-server` and `dashboard-opcua-client` images in Docker 27.4.0-rc.3 on Debian 12 VM.

**Log**
```
2024-12-04 09:52:32,448 [DashboardOpcUaClient] INFO std::shared_ptr<ModelOpcUa::StructureBiNode> Umati::Dashboard::OpcUaTypeReader::handleBrowseTypeResult(BiDirTypeMap_t&, const ModelOpcUa::BrowseResult_t&, const std::weak_ptr<ModelOpcUa::StructureBiNode>&, ModelOpcUa::ModellingRule_t, bool):255 Current size BiDirectionalTypeMap: 500
[2024-12-04 09:53:18.750 (UTC+0000)] error/client       The server certificate is different from the EndpointDescription
[2024-12-04 09:53:18.750 (UTC+0000)] error/client       Processing the message returned the error code BadSecurityChecksFailed
[2024-12-04 09:53:18.750 (UTC+0000)] info/channel       TCP 6   | SC 7470       | SecureChannel closed
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        Received Publish Response for a non-existant subscription
[2024-12-04 09:53:18.750 (UTC+0000)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-12-04 09:53:18.754 (UTC+0000)] info/network       TCP 8   | Opening a connection to "server-cpp" on port 4840
[2024-12-04 09:53:18.754 (UTC+0000)] info/client        Client Status: ChannelState: Connecting, SessionState: Created, ConnectStatus: Good
2024-12-04 09:53:18,754 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):39 Client is connecting
2024-12-04 09:53:18,754 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):78 Session created
[2024-12-04 09:53:18.754 (UTC+0000)] info/network       TCP 6   | Socket closed
2024-12-04 09:53:18,754 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):45 Client hello sent
2024-12-04 09:53:18,754 [DashboardOpcUaClient] INFO void Umati::OpcUa::stateCallback(UA_Client*, UA_SecureChannelState, UA_SessionState, UA_StatusCode):78 Session created
2024-12-04 09:53:18,754 [DashboardOpcUaClient] ERROR void Umati::OpcUa::OpcUaClient::connectionStatusChanged(UA_Int32, UA_ServerState):686 Disconnected.

terminate called after throwing an instance of 'Umati::Exceptions::ClientNotConnected'
  what():  Need connected client.
```

GoetzGoerisch, 2024-12-06:
@ccvca can you have a look please.
@ainglessi you have a suspected commit from upstream, could you add this here?

wlkrm, 2025-01-31:
@ainglessi  is this solved?
https://github.com/open62541/open62541/issues/7043

ainglessi, 2025-01-31:
The line

`error/client       The server certificate is different from the EndpointDescription`

is not there any more, but

`error/client       Processing the message returned the error code BadSecurityChecksFailed`

is still there and the gateway still crashes.

May be another issue.

wlkrm, 2025-01-31:
Ah ok will look into that

ainglessi, 2025-02-13:
I've been running revert-431 with the latest server for a couple of days now, the error hasn't popped up.

wlkrm, 2025-02-13:
Ok, would you considered this bug as "fixed" then?

ainglessi, 2025-02-13:
I think we can close it, keeping in mind that it may be reopened in case the error can be reproduced again.

ainglessi, 2025-02-13:
Can not reproduce reliably.

closed 2025-02-13 by ainglessi