umati/Dashboard-OPCUA-Client #912: Make Client ready for archieving

wlkrm, 2025-01-21:
v.2.0.0-rc4 before the TypeDictonary merge #431  was the last stable version before the client started to break.
Known sources of the accumulated issues:

- TypeDictonary (#431)
- Dynamic Jobs (#129), cannot find corresponding pull-request
 
Issues to fix forch achiving: 
- [ ] #868 
- [ ] #698 
- [ ] Reconnection Logic
- [ ] ResultManagement in GMS

wlkrm, 2025-01-21:
@GoetzGoerisch @ainglessi please add details, other known issues, etc.
@mdornaus @Kantiran91 please describe, which commits after v.2.0.0-rc4 / the ModelChangedEvent are needed.

ainglessi, 2025-01-22:
Just checked #868, still happens with the latest SampleServer and Client images.

wlkrm, 2025-01-23:
@ainglessi, does #868 work with v.2.0.0-rc4 ?

ainglessi, 2025-01-23:
> [@ainglessi](https://github.com/ainglessi), does [#868](https://github.com/umati/Dashboard-OPCUA-Client/issues/868) work with v.2.0.0-rc4 ?

Looks like it, no crashes within half an hour so far. Trying `sample-server:develop` with `dashboard-opcua-client:v2.0.0-rc4`.

wlkrm, 2025-01-24:
I suggest that we roll back to v.2.0.0-rc4. Add the important bugfixes from development branch and replace the
type dictionary by manual decoding of the ResultType.

Kantiran91, 2025-01-28:
@wlkrm I think  the Feature with the version is interesting. The rest im not sure.

wlkrm, 2025-02-11:
ResultType is fixed, as far as the ResultMetaDataType and the "HasStructeredComponet" version of the encoding is concerned:
```json
    "Results": {
      "<ResultVariable>": {
        "Result": {
          "$TypeDefinition": "nsu=http://opcfoundation.org/UA/Machinery/Result/;i=2001",
          "properties": {
            "ResultMetaData": {
              "properties": {
                "FileFormat": [
                  "PDF"
                ],
                "ResultId": "-",
                "ResultState": 1
              },
              "value": {
                "FileFormat": [
                  "PDF"
                ],
                "ResultId": "-",
                "ResultState": 1,
                "ResultUri": [
                  "http://www.wenzel-group.com/result"
                ]
              }
            }
          },
          "value": {
            "ResultMetaData": {
              "ResultId": ""
            }
          }
        }
      }
    }
  }
}
```

closed 2025-03-12 by wlkrm