umati/Dashboard-OPCUA-Client #915: LaserSystem crashes the gateway

ainglessi, 2025-01-23:
**Describe the bug**
When trying to add LaserSystem type and connect to the Node sample server, the gateway crashes.

**To Reproduce**
Steps to reproduce the behavior:

* Include `"http://opcfoundation.org/UA/LaserSystems/"` in "ObjectTypeNamespaces".
* Add this to "NamespaceInformations":

```
      {
      "Namespace": "http://opcfoundation.org/UA/LaserSystems/",
      "Types": [
        {
          "Uri": "http://opcfoundation.org/UA/LaserSystems/",
          "Id": "i=1005",
          "$comment": "LaserSystemType"
        }
      ],
      "IdentificationType": {
        "Uri": "http://opcfoundation.org/UA/LaserSystems/",
        "Id": "i=5004",
        "$comment": "MachineryComponentIdentificationType"
      }
    }
```

* Run `dashboard-opcua-client:v2.0.0-rc4` image against `sample-server-node-opcua:develop` image.
* Experience crash (see log below).

**Expected behavior**
No crashes, machine shows up in the dashboard.

**Environment:**

VM, Docker

**Log**

```
gateway-dev-pr  | terminate called after throwing an instance of 'Umati::Exceptions::UmatiException'
gateway-dev-pr  |   what():  Invalid NodeClass
gateway-dev-pr  | 2025-01-23 15:12:42,455 [DashboardOpcUaClient] ERROR UA_NodeClass Umati::OpcUa::OpcUaClient::nodeClassFromNodeId(const open62541Cpp::UA_NodeId&):514 Invalid NodeClass 1 expect object or variable type for node
gateway-dev-pr exited with code 0
```

**Additional context**

Latest `dashboard-opcua-client:development` image crashes with this error (with and without LaserSystems added):

```
gateway-dev-pr  | terminate called after throwing an instance of 'Umati::Exceptions::OpcUaNonGoodStatusCodeException'
gateway-dev-pr  |   what():  806f0000, BadNoMatch
gateway-dev-pr exited with code 139
```

ainglessi, 2025-01-24:
Currently I'm having troubles running the latest umatiGateway. Working with @mdornaus to make it work.

Kantiran91, 2025-01-27:
It tested it today with the latest Version of the umatiGateway and that works. (see https://dev.umati.app/all/LaserSystemType/clientId%3Dfva%7Ctest%3Bnsu%3Dhttp%3A_2F_2Fvdma.org_2FUA_2FLaserSystem-Example_2F%3Bi%3D5003)

ainglessi, 2025-01-27:
Can confirm that it works with umatiGateway. @GoetzGoerisch, shall we leave LaserSystems unsupported in the old client?

ainglessi, 2025-01-29:
Closing due to shifting focus on the new gateway.

closed 2025-01-29 by ainglessi